NIURoverTeam/RoverCoreOS #3: Create URDF File for Ad Astra

LuckierDodge, 2019-02-28:
http://wiki.ros.org/urdf/Tutorials

adugyan, 2019-03-03:
> Assigned #3 <https://github.com/NIURoverTeam/RoverCoreOS/issues/3> to
> @adugyan <https://github.com/adugyan>.

Thanks. I’ll be working on it today

adugyan, 2019-03-13:
What text editor and file type should I be using for the urdf? I can’t
figure out how to launch this

LuckierDodge, 2019-03-14:
I'd recommend running through some of the tutorials for URDF [here](http://wiki.ros.org/urdf/Tutorials), to get a better understanding of what URDF files are and how you can edit and launch. To actually view them, you're gonna have to be booted into Linux.

LuckierDodge, 2019-04-19:
[STL FILES.zip](https://github.com/NIURoverTeam/RoverCoreOS/files/3096684/STL.FILES.zip)

closed 2019-10-04 by LuckierDodge