NIURoverTeam/RoverCoreOS #4: Update Documentation and Getting Started Guide

LuckierDodge, 2019-02-28:
Need to update docs to reflect changes to project, including switch from Lunar to Kinetic, use of VMs versus Dual Boot versus WSL, and list of courses to learn things like python, command line/Linux, etc.

LuckierDodge, 2019-03-16:
Thing that needs to be added: New users for TX2 or Shatterdome need to be added to the sudo, dialout, tty, and i2c groups in order to be able to run all of the various aspects of the package. Logout required after adding for effects to manifest.

waffleston, 2019-03-27:
The Wiki has been set up, and the two files in the `/docs/` folder were imported into it.  Should I remove the `/docs/` folder from the repository?

LuckierDodge, 2019-03-27:
Yeah, no need for duplicates

LuckierDodge, 2019-03-27:
Just took a look at the wiki, it's a great start! Definitely think it'll work better than just having a Doc directory.

LuckierDodge, 2019-06-25:
The bulk of this work is done. Last bits can be fleshed out as needed, as part of the Crash Course development.

closed 2019-06-25 by LuckierDodge