NIURoverTeam/RoverCoreOS #8: Create a Crash Course Curriculum

LuckierDodge, 2019-03-27:
Need to build out a set of beginner's guides to bring new developers up to speed as quickly as possible.

Things to address:
- [ ] Bash and Linux Development
- [ ] Git and GitHub
- [ ] Arduino?
- [ ] ROS Tutorials Guide
- [ ] Python
- [ ] Development Practices
- [ ] Robotics Example Project

Chime in with anything else ya'll think we need to address.

adugyan, 2019-05-20:
https://www.robotigniteacademy.com/en/path/ros-for-beginners/

LuckierDodge, 2019-05-25:
Git & GitHub: https://guides.github.com/, specifically the **Understanding the GitHub Flow** and **Hello World** tutorials.

adugyan, 2019-07-02:
https://github.com/adugyan/Bash-and-Linux-Development-/wiki

LuckierDodge, 2019-07-02:
@adugyan can you move that into the wiki and add a link to it on the Wiki Home Page under Crash Course? Thanks!

adugyan, 2019-07-03:
Yea, I'll do that right now

closed 2019-10-04 by LuckierDodge