NIURoverTeam/RoverCoreOS #16: Test 3D Mouse

LuckierDodge, 2019-10-05:
Try to get the 3D Mouse up and running. The ROS node for the one we have is here: http://wiki.ros.org/spacenav_node.

Primarily concerned with getting either the /joy or /twist topics

DrakeProvost, 2019-10-15:
- followed link that Ryan posted
- when doing the 2 installations, make sure to switch `indigo` to your current distro (e.g. melodic)

DrakeProvost, 2019-10-15:
- ssh'ed into shatterdome, and then installed the spacenav stuff there

DrakeProvost, 2019-10-15:
- set up ROS on my user account on shatterdome using bits from this tutorial: http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment
- ran `roslaunch spacenav_node classic.launch`
- opened a new terminal
- to view any of the 4 topics described in the [spacenav_node package summary](http://wiki.ros.org/spacenav_node), use `rostopic echo /topic_name` to view messages published to the topic, and `rostopic echo -c /topic_name` to view messages published and have the screen cleared between each update, which is more readable

DrakeProvost, 2019-10-16:
Topics tested with notes about each one:
- `spacenav/offset`: The package summary says `Publishes the linear component of the joystick's position. Approximately normalized to a range of -1 to 1`, but the range seems to be less than that. **All 3 linear axes are ±0.68359375**
- `spacenav/rot_offset`: Same deal as spacenav/offset. The package summary says `Publishes the angular component of the joystick's position. Approximately normalized to a range of -1 to 1.`, but the ranges for **all 3 rotational axes are ±0.68359375**
- `spacenav/twist`: Package summary says it `combines offset and rot_offset into a single message`, it does in fact do that
- `spacenav/joy`: The stuff in the header is the `time the data is received from the joystick`, according to the package summary's documentation. For the axes, they are as follows: `[linear x, linear y, linear z, angular x, angular y, angular z]`. If holding the 3D mouse so the cord is pointing toward your body and the logo is pointing away, then the buttons are as follows: `[right button, left button]`. Everything is publishing successfully, so the 3D mouse works. 🎉

LuckierDodge, 2019-10-16:
So, one thing to note here: "normalized" in this context is probably referring to the magnitude of the total vector. i.e. the maximum length of the vector defined by <x, y, z> is approximately <= 1. Remember that you can move along multiple dimensions simultaneously. Same for rotation.

DrakeProvost, 2019-10-16:
Ah ok, that makes sense. Those upper and lower bounds still seem to hold true for any 1 axis because they seem to be the physical limits of the mouse (like, pushing or pulling the z axis to its fullest extent would give the +-0.6 ish number regardless of the x and y, and this seemed to be the case for all 6 degrees of freedom), but further testing is always a good idea so we can better understand how it defines each dimension when they're not at the high or low bounds. Thanks for the note!

LuckierDodge, 2019-10-20:
@DrakeProvost sometime this week can you writeup a doc page in the wiki with this info? Installation, messages used and supported, any tips/tricks etc.

DrakeProvost, 2019-10-20:
Sure thing. ??

closed 2019-11-18 by LuckierDodge