NIURoverTeam/RoverCoreOS #18: Encoders and Feedback

LuckierDodge, 2019-10-05:
Need to figure out what types of encoders we can get for the motors and actuators controlling the arm, and how we can communicate with them. http://wiki.ros.org/ros_control might(?) be helpful for this.

DrakeProvost, 2019-10-16:
"We're going to be using the same actuators that are on the arm currently, as well as one of the other actuators that we have. We have 2x 6 inch actuators and 1x 10 inch, and while neither of them have encoders inside of them, so we're going to make a system with a gear that measures the distance the arm is extended, but as for motors I don't have any specifics just yet, but the ones that are currently mounted on to the arm are a good basis" -Spenser (arm team lead), 10/16
"Alrighty, so as of right now I have no data on the arm so I can't give information on that until they give me a information about the arm... We will talk to Spenser tomorrow and see whats going on with it. Also if not tomorrow give me till Tuesday because I'm having us push to have the Rover ready on Saturday." - Nick (ELE team lead), 10/16

I'll return to this issue in a few weeks to give the 2 teams some more time to solidify their plans.

LuckierDodge, 2019-10-30:
Turns out we've had encoders the whole time!

LuckierDodge, 2019-11-18:
Encoders are currently sitting on the big table in the middle of the Rover space. @NickELE  will hopefully get those hooked up soon, at least to a dummy wheel, so we can start testing this with sensor fusion for odometry.