NIURoverTeam/RoverCoreOS #19: URDF Models

LuckierDodge, 2019-10-05:
1. Bring model of the Rover 5 from old repo into new one. Get it up and running and make sure it still works.
2. Model of the Rover itself. Either make a crude approximation from simple geometry, or get .stl files from mechanical.
3. Model of the arm. Can be pretty simple, primarily just need it for demo purposes.

DrakeProvost, 2019-10-31:
I'm currently in the works of obtaining CAD files from the chassis team so I can convert them to URDF files. If you were a sane person you'd use the [Solidworks to URDF Exporter add-in](https://wiki.ros.org/sw_urdf_exporter) to automatically convert Solidworks files into the URDF models necessary for this issue. I don't have solidworks installed though and I don't wanna go through the hassle of obtaining and installing that from NIU Robotics, so I'm gonna do a bit of a bootstrap thing instead and hope it works. 

My plan is such:

1. Make a chassis person save their assembly and its parts as a .zip file using the ["Pack and Go" feature](https://www.fisherunitech.com/blog/always-complete-packing-list-solidworks-pack-go) and then email that to me.
2. Import the .zip file into [Onshape](https://www.onshape.com/), a CAD program that runs in the browser and uses cloud storage to save files 🤓. It sucks to use for actual CAD, but it's pretty nice if you want to do CAD things but don't want to go through the hassle of installing a CAD software like I'm doing now.
3. Export the assembly in, wait for it, collada format (.dae extension). I picked this format because it was the only format I could export from Onshape that I could find a "convert to URDF" tool for.
4. ROS has a package called [collada_urdf](http://wiki.ros.org/collada_urdf#collada_to_urdf_command-line_tool%3C/p%3E). We can use `rosrun collada_urdf collada_to_urdf <input.dae> <output.urdf>` to convert our collada file to URDF now.

This plan is either big brain or thick skull, but I hope it works either way. Wish me luck.

LuckierDodge, 2019-10-31:
Outstanding! The workflow we need, but not the one we deserve.

DrakeProvost, 2019-11-13:
Notes I learned about my process I described:

- When asking a chassis person for the .zip file, make sure to request that they name the file the same name as the root .SLDASM file. Onshape uses the name of the .zip file to determine which .SLDASM file to make as the root, so if you don't mention this to the chassis person, they might rename it `Rover.zip` and then you'll have to either ask or unzip it, skim through the couple hundred files, and take a lucky guess on which .SLDASM file is the root one. `New Rover 2.08 12.18.18.zip` was the right one for me, but it's easier if you just skip this ordeal and ask them to name it the same as the root assembly file.

DrakeProvost, 2019-11-15:
Sad news--I don't think this collada_urdf package is worth the extra hassle. It was a pain to get CAD files in the correct format for Onshape to accept the import, it took me like 7 tries to export it without my anti-virus software blocking the download (which was obnoxious to sort out), I had to google a lot to figure out I had to `sudo apt-get install ros-<distro>-collada-urdf` since the collada_urdf package documentation does not say anything about it, and now when I just did ` rosrun collada_urdf collada_to_urdf New-Rover-v2_08-12_18_18.dae Rover-2018dec18.urdf`, it gave back: 
```;; Input file is: New-Rover-v2_08-12_18_18.dae
Error:   Error document empty.
         at line 71 in /build/urdfdom-ACOcA8/urdfdom-1.0.0/urdf_parser/src/model.cpp
ERROR: Model Parsing the xml failed```

At this point I'm running against a sunk cost fallacy and I think I'm just wasting my time with something that is more hassle than it's worth, so I guess I'm gonna bite the bullet and install solidworks. This package was worth a shot, but it seems to be a bigger ordeal than I was hoping and I think I should just move on to something more likely to work.

DrakeProvost, 2020-01-22:
Currently have done 2 attempts of making the URDF model with the SolidWorks URDF exporter. The first attempt (which you can find in the commit history if you'd like) has 0 joints if I remember correctly so it's basically just a really fancy brick, and the second attempt has around 7 or so joints at places I thought they belonged, but I could be wrong with how I set them up. At this point, the next steps would be to:
1. Test rviz on shatterdome with the rover 5 URDF model from last year, since I assume that model is more likely to be functional than the models I just made
2. Test either or both of the ad astra URDF models I committed to see if they work as expected/desired
3. Remake the URDF model if necessary