NIURoverTeam/RoverCoreOS #24: Gazebo Simulation Example

LuckierDodge, 2019-10-05:
https://wiki.ros.org/gazebo_ros_pkgs

Want to get a working example of this running, either on a laptop or on the base station. Can just be a simulation of the Rover5 at this point. Also need to determine if we need to order a Graphics card to support this.

sam-thompson1752, 2019-10-16:
Gazebo is up and running. It took a simple sudo apt upgrade to make gazebo 9 work.

sam-thompson1752, 2019-10-16:
RVIS seems to be acting up. Is giving me segmentation fault (core dumped) errors every time I try to run it. I tried upgrading OpenGL, to no avail. The troubleshoot page hasn't helped either. I will continue to try to fix this, and if not, I will see if RVIS is even necessary. Will comment with more updates in the future.