NIURoverTeam/RoverCoreOS #30: IP Managment Utility

LuckierDodge, 2019-10-20:
In order to use our antennas, we need a static IP address. But we also need to be able to connect to the internet when not using the antennas. So we need to be able to quickly and easily switch between the two. Here's what I'm thinking:

A bash script, located in `RoverCoreOS/util/`:

Usage:
```
ip_man <-d/s> [ADDRESS]
```

* `-d`: switch to DHCP (normal) mode
* `-s`: switch to a static IP specified by `ADDRESS`

waffleston, 2019-10-31:
Experimented with creating such a script, found it quirky.

Opted to try using a router instead.  Selected the "ASUS AC750" from the CS cabinet.

That router has been configured to assign addresses via DHCP except for certain MAC addresses.  This was tested with my laptop and appears to work well enough.

Current tasks include determining how many devices will need static IP addresses, and determining the MAC addresses of such devices.

LuckierDodge, 2019-11-18:
Right now, devices needing static ip addresses are as follows:

* Shatterdome
* The Jetson TX2
* Jetson Nano 1
* Jetson Nano 2

We'll come up with clever host names at some point for the latter 3.